apollo-lhc/cm_mcu #39: cmpwrup fails from time to time

rzouCERN, 2020-02-06:
**Describe the bug**
When commands cmpwrdown cmpwerup are called, from time to time cmpwerup will fail with the message:
CM 1 failed to powered up in time

And PWR_GOOD appears to be 0.

**To Reproduce**
1. Log in cms@apollo
2. do 'BUTool.exe -a /opt/address_tables/connections.xml'
3. do 'cmpwerdown' cmpwerup' multiple times until error appears

pwittich, 2020-02-07:
[Possible workaround: increase this delay](https://github.com/apollo-lhc/cm_mcu/blob/361817cbb15b5c7018db1e2ad0995f5fec1d5283/common/power_ctl.c#L286)

My working hypothesis is that the issue would be resolved by implementing the fix in #26

pwittich, 2020-05-26:
Can't reproduce. Presumed solved by #51 (these were system stack corruption issues.)

closed 2020-05-26 by pwittich