apollo-lhc/cm_mcu #50: crash in errbuff_out

pwittich, 2020-04-03:
https://github.com/apollo-lhc/cm_mcu/blob/c50dedba46139e610cd0c5890607757409799cc9/projects/cm_mcu/CommandLineTask.c#L943-L951

I think there are two issues in this function
1. here you are using a somewhat obscure feature of c called "variable length arrays" that depends on the stack size being big enough. On our little microcontroller it is probably safer to instead create a static array with some size (like 25 entries) and then if the argument asks for more than that limit the output to 25 entries and print an error message 
1. The buffer we write to `m` overruns its boundary when `copied>SCRATCH_SIZE` if you look at e.g. `adc_ctl` we handle this by returning `pdTRUE` when we are close to overrunning the buffer; the CLI tool will keep calling the function until all values have been copied.  The signaling mechanism is the static variable `whichadc` which keeps track of where you are in the output loop.
https://github.com/apollo-lhc/cm_mcu/blob/c50dedba46139e610cd0c5890607757409799cc9/projects/cm_mcu/CommandLineTask.c#L507-L527

gwengardner, 2020-04-21:
I've made the two changes to errbuff_out, and pushed the changes to the branch errbuffer_messages. I see that some of the other functions don't all check for `copied>SCRATCH_SIZE`, but it seems that all of those are short enough that they won't cause issues unless we add several lines. Should I go ahead and throw in an if statement anyway?

pwittich, 2020-04-21:
I guess to be safe that would be prudent, though there are a lot of functions in that file that also should be updated the same way ....

pwittich, 2020-05-05:
#53 closes this issue

closed 2020-05-05 by pwittich