apollo-lhc/cm_mcu #117: fpga commands on MCU not working

pwittich, 2021-11-18:
**Describe the bug**
`fpga done` does not appear to show the correct state of the done bit
`reset_fpga k` does not reset f1 as expected.

pwittich, 2021-11-18:
CRS says FPGA_DONE is bad due to Rev2 hardware issue

pwittich, 2021-11-19:
@pkotamnives @rzouCERN can one of you reset a Rev1 fpga next time you've programmed one and confirm it works?

pkotamnives, 2021-11-19:
@pwittich do you mean that just `reboot` the zynq and check if after programming the fpgas they are all detectable by `status 3 FIRMWARE`?  I can check that now if so.

pwittich, 2021-11-19:
no, I mean checking `status 3 FIRMWARE` before and after doing `reset_fpga k` or `reset_fpga v` on the microcontroller command line. I think you can do that form BUTool via `uart_cmd CM_1 reset_fpga k`

pwittich, 2021-11-22:
@pkotamnives confirms this works on Rev1.

pwittich, 2023-06-02:
still doesn't appear to work

closed 2023-06-06 by pwittich