apollo-lhc/cm_mcu #124: Store clock chip information on Rev2b eeprom and use it to initialize clocks

pwittich, 2022-02-17:
We need to store the programs for initializing the clocks on the board so that the clocks are loaded with default programs by default.

This issue depends on issue #123 being completed.

At MCU reboot and when prompted via the CLI (and possibly via a register write) the clock chips will need to be programmed via I2C by the MCU. So we need to prepare for this.

Steps
1. generate the data for the various clocking chips on the CM. (I think there are 5?) I think @CRStrohman or @rzouCERN need to do this
2. program the data into the bigger external EEPROM on Rev2b
3. create a function in the MCU that will read the data from the EEPROM and write it to the appropriate SiLabs chip
4. Add this function to the initialization task after here https://github.com/apollo-lhc/cm_mcu/blob/4479f34eda6a9a73dfcfe26c54713109c439215f/projects/cm_mcu/InitTask.c#L36
5. add a CLI interface to program the clocks

We might need more than one program for each clock interface, so we should allow for this both in how the data is stored in the EEPROM, how it is loaded by the MCU, and in the CLI interface

pwittich, 2022-02-21:
some comments from Charlie 

> On the board in my basement, I replaced the EEPROM 
> with a 24CS512 (datasheet attached). I'll replace it in the board in the 
> lab soon.
> 
> It has 64k bytes, organized as 512 pages of 128 bytes. Random access to 
> any address requires 4 I2C operations (device, high address, low 
> address, data). Sequential operation that only requires the initial 
> device and address is supported within a page.
> 
> The EEPROM is on the same I2C bus as the clock chips, but is before the 
> bus switch. You can set the switch for the chip that you are 
> configuring, then read EEPROM data and write it to a clock chip without 
> fussing with the switch.
> 
> One thought about storing clock chip configuration data is:
> 
> a) for each of the 5 clock chips, store the preamble in one page, the 
> data in a series of 14 pages, and the postamble in another page.
> 
> b) In the clock chip configuration files, each register has a 2-byte 
> address and 1-byte data. The high-byte of the address is the page, and 
> the low byte is the address within the page. The page is changed by 
> writing to address 0x01 on any page. I'd propose that we pre-process the 
> data so that the page changes are embedded in the stream. This way, each 
> register only needs to have 2 bytes stored. One address byte and one 
> data byte.
> 
> c) The clock chips have on the order of 600 registers and require about 
> a dozen page changes, so 14 pages of 128 bytes could store 869 register 
> or page-change instructions. We should look at the performance to 
> determine if we want to store and write all register data, or only the 
> registers that are known to be different from the power-up defaults.
> 
> d) Utilize sequential reads from the EEPROM to access each page of the 
> clock chip data.
> 
> e) consider a header table that points to the EEPROM starting address 
> and number of words required to configure each clock chip, including 
> pre- and post- ambles.

pkotamnives, 2022-04-07:
Hi @pwittich, 
The latest commit to `read_eeprom` [here](https://github.com/apollo-lhc/cm_mcu/commit/6e41058618c0abeb8a5de8c103ee0c076fd19ca6) has been tested for programing clock chips with id r0a and r1a successfully only if the configCHECK_FOR_STACK_OVERFLOW is set to 0. Otherwise, the program got stuck by stack overflow.

pwittich, 2022-04-07:
If you set the check to zero, it doesn’t mean you don’t have a stack overflow — it just means you didn’t detect it and Bad Things will happen later….  Can you post the error you are getting? (There are two stacks on the Cortex-M3, the task stack and the system stack, I would like to know which is overflowing.)

pwittich, 2022-04-07:
Also we need to think how to check if the programming was successful (test the clock output)

pkotamnives, 2022-04-07:
re: which stack. It was this line that got called and frozen `https://github.com/apollo-lhc/cm_mcu/blob/read_eeprom/projects/cm_mcu/cm_mcu.c#L401`.

re: test clock output. Do you mean the output such as frequency and phases of clocks after being programmed? aside from this, do we also want a part in the code that throws an error if data written to and read back from the clock are different (as I have checked for this code)? I just think that as long as the statuses from i2c write/read success they should be the same except 0x001C register for example.

pwittich, 2022-04-07:
Ok, what was the value of `pcTaskName`?

Re output, yes, I want to make sure that the output is the right one (mainly clock frequency.) The proof is in the pudding, as they say.

pkotamnives, 2022-04-07:
pcTaskName was INIT.

pwittich, 2022-04-07:
Can you post the entire output of the MCU from boot til stack overflow.

pkotamnives, 2022-04-07:
I built and run the code that I just committed in which the stack overflow check turned on with value of 2 but I can't reproduce the stack oveflow I saw yesterday. It runs successfully today in debugger (boot happened earlier in the screen) and in CLI:

```
CLI based on microrl.
Type "help" to view a list of registered commands.
% 8 SRV INF cm_mcu.c:426:Stack canary now 52
14 SRV INF InitTask.c:36:Clock I/O expander initialized
32 SRV INF LedTask.c:133:stack (LED) = 56(was 217)
50 SRV INF MonitorTask.c:265:stack (XIMON) = 428(was 4096)
75 SRV INF MonitorTask.c:265:stack (XIMON) = 274(was 428)
98 SRV INF PowerSupplyTask.c:372:stack (POW) = 296(was 4096)
100 PWR INF PowerSupplyTask.c:222:power-up requested 
100 SRV INF ADCMonitorTask.c:311:stack (ADC) = 16(was 169) 
MON(XIMON): 3V3 came back. Restarting I2C monitoring. 
125 MON INF MonitorTask.c:130:XI
MON: PWR on. (Re)starting I2C monitoring. 
MON(XIMON): Page SMBUS ERROR: 5 
MON(XIMON): Error 5, break out of loop (ps=0,c=0,p=0) ... 
MON(XIMON): Page SMBUS ERROR: 5 
MON(XIMON): Error 5, break out of loop (ps=1,c=0,p=0) ...
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=2,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=3,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=4,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=5,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=6,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=7,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=0,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=1,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=2,c=0,p=0) ... 
 MON(XIMON): Page SMBUS ERROR: 5 
 MON(XIMON): Error 5, break out of loop (ps=3,c=0,p=0) ...
  --> suppressing further errors for now 187 
  MON WRN MonitorTask.c:70:suppressing further errors for now 
  200 SRV INF MonitorTask.c:265:stack (XIMON) = 264(was 274) 
  269 FFL INF FireFlyTask.c:587:initialization complete. 
  503 SRV INF ZynqMonTask.c:440:stack (ZMON) = 12(was 135) 
  677 SRV INF MonitorTask.c:265:stack (PSMON) = 135(was 4096) 
  1283 SRV INF MonitorTask.c:265:stack (PSMON) = 18(was 135) 
  7577 SRV INF InitTask.c:40:Configuring clocks initialized % 
  13908 SRV INF CommandLineTask.c:885:stack (CLIZY) = 369(was 4096) % 
  13927 SRV INF CommandLineTask.c:885:stack (CLIZY) = 232(was 369) 
  % loadclock 1 
  loadclock is programming clock r0a. 
  clock synthesizer with id r0a successfully programmed. 
  % loadclock 3 loadclock is programming clock r1a. 
  clock synthesizer with id r1a successfully programmed. 
  %
```

closed 2022-05-05 by pkotamnives