apollo-lhc/cm_mcu #126: Monitor the status registers of the SI chips

rzouCERN, 2022-03-08:
**Is your feature request related to a problem? Please describe.**
SI5341 and SI5395 chips have status registers (LOS, OOF, etc) that we should monitor when the chips are being used.

For SI5341: see Table 4.5 : [https://www.skyworksinc.com/-/media/Skyworks/SL/documents/public/reference-manuals/Si5341-40-D-RM.pdf](https://www.skyworksinc.com/-/media/Skyworks/SL/documents/public/reference-manuals/Si5341-40-D-RM.pdf)

For SI5395: see Section 5.3: [https://www.skyworksinc.com/-/media/SkyWorks/SL/documents/public/reference-manuals/si5395-94-92-family.pdf](https://www.skyworksinc.com/-/media/SkyWorks/SL/documents/public/reference-manuals/si5395-94-92-family.pdf)

For the sticky flags we should also write 0s to them at the end of programming.

pwittich, 2022-03-22:
We could model the monitoring after the MonitorTask and possible adapt the Firefly and this new task to use the same code (similar to how the MonitorTask is used for both Xilinx and PS.) 
It might be possible to immediately reuse that code if the clock chips speak the I2C extension called SMBus / PMBus.

@rzouCERN have you by any chance in your testing to date already read out these registers (just to confirm we are able to do so)?

pkotamnives, 2022-08-03:
Hi @pwittich and @rzouCERN , the clock monitoring instance in the new MONI2C task seems to work just fine along with clock loading now. I would like to know if Rui means reading all the bits of the register addresses in table 4.5 for SI5341 to monitor 12 status names or just read one byte of each register address shown in the table (so the total status names would be 4)

Also, should I still write 0s to the register addresses 0x11 and 0x12 which are sticky flags of SI5341 and SI5395 ? I notice that the SI5295 does not have the sticky OFF flag. Why is that the case? but it's needed to be 0 at the end of clock monitoring as well, correct?

pwittich, 2022-08-04:
@pkotamnives I would propose we do not clear anything in the monitoring, otherwise we will miss transient errors. The monitoring should (IMO) be read-only. I would propose the following.

For Si5341: Read registers at 0xC and 0xD

For Si5395: Read registers at 0xC[1] and 0xD [0:3], 0xD[7:4], 0xE[1]

I think we should just store send them to grafana as hex numbers, though I have not yet found a good way to display this information in Grafana (maybe you have, @rzouCERN ?)

pwittich, 2022-08-22:
closed by #136

closed 2022-08-22 by pwittich