apollo-lhc/cm_mcu #140: handle 3.8V for 25G Firefly

pwittich, 2022-08-22:
We need to provide an easier way to turn on the 3.V for the fireflies, and it also needs to be turned off before the 3.3V is turned off.

pwittich, 2022-09-01:
To do
- write a function for manipulating the relevant I/O expander to turn it on
- add CLI interface to this function 
- add turning on the 3.8V to the power sequencing (and monitoring?) in `PowerSupplyTask`. 

This is not just flipping an enable bit so it is a bit complicated/different than the other power supplies.

CRStrohman, 2022-09-01:
Should we use the “present" signal to figure out which FFs are installed? Then the mask to determine which installed devices to ignore?

Once we know which are in, should we query them to determine which are 25G x12 transmitters? I assume that can be done? Only 6 of the 20 slots can have these. Maybe then read the switch positions, and flag an error if the position is wrong, either in the 3.8V position for a non-25G x12 or in the 3.3V position for a 25G x12. Finally enable 3.8 only for populated transmitters that are 25G x12.

Charlie

pwittich, 2022-09-02:
@CRStrohman In our last discussion on this topic I thought we had decided that there is no harm in turning on the 3.8V even if the CERN-B parts are installed (since there is a physical jumper that decides whether the part gets 3.3V or 3.8V) that we would simplify the programming by always turning on the 3.8V.  Is that not a good idea?

pwittich, 2022-09-02:
The mask question is being addressed in #139 by @pkotamnives — she will program the masks based on the “present” signals on boot and also set the mask in the EEPROM on first boot.

CRStrohman, 2022-09-02:
There is no harm turning on the 3.8v so long as the switch is in the 3.3v position. If the switch is inadvertently in the 3.8v position then the 3.3 and 3.8 will be shorted together if a CERN-B part is installed.

pwittich, 2022-10-21:
#153 added the CLI.
remaining issue now is to hook it into the power-up and power-down sequence, and to decide how to ensure that we don't turn on the 3.8V when we shouldn't.

pkotamnives, 2023-03-19:
> #153 added the CLI. remaining issue now is to hook it into the power-up and power-down sequence, and to decide how to ensure that we don't turn on the 3.8V when we shouldn't.

Could you confirm again what should be a sequence to turn on the 3.8V for supplying power to boards with 25Gb x 12-ch? should we discuss this again with Charlie again but he seems to suggest that we should turn 3.8V by step 6?

CRStrohman, 2023-03-19:
Turning the 3.8 volts OFF in step 6 seems to make sense, so that the 3.8 
goes away before the 3.3. I still think I'll put some diodes on to cover 
the case where board power goes away without a clean shutdown. Turning 
the 3.8 ON in step 6 keeps things logical.

I can change my diagram when this is done.

Charlie

PR #183 (2023-05-19): Add a power control state machine state 6 for controlling V3.8 needed for Samtec 12-ch 25Gbs
This PR will resolve #140 

Note that this change is only needed for REV2. Currently, the infrastructure of this state machine is ready to be reviewed. We only need to add a bit mask to ff struct such as `ffl12_f1_args`  and `ffl12_f2_args` to store information if a vendor part of each ch is a 25Gb one to enable V3.8, and the state machine condition.
closed 2023-05-19 by pwittich