apollo-lhc/cm_mcu #150: ff xmit on/off all doesn't work as expected

pkotamnives, 2022-10-14:
issue : @rzouCERN noticed a weird behavior of `ff xmit on all` and `ff xmit off all` commands with REV2 board (CM 203) on Apollo205. The current version is 
```
% version
Version commit date 2022-09-20 22:51:22 -0400   (HEAD -> hotfix/rev1, origin/hotfix/rev1) v0.99.1-18-g823cacb-dirty
regular build 
built at 15:49:59, Sep 21 2022.
```
basically, the FFs seem to always be disabled. That could be explained by [this line](https://github.com/apollo-lhc/cm_mcu/blob/master/projects/cm_mcu/commands/SensorControl.c#L878) in the code (also for other places) that also exists in the master branch. This seems to skip the line(s) with "on" by exiting the scope too soon.  

solution : fix this bug. However, even if this bug is fixed, I am still not sure what would be an onset of commands to test. I have tried writing/reading from the register 0x34 (Transmit Channel Disables) directly but got confused by the output that I would expect from turning on disabled transmit link: 
```
% i2cwr 4 0x50 1 34  2 0                                                                                                
i2cwr: W to addr 0x50, reg 0x34, val=0x00000000 (2 bytes)                                                               
% i2crr 4 0x50 1 34 2                                                                                                   
i2cr: add: 0x50, reg 0x34: value 0x00000000 (2 bytes)                                                                   
% i2cwr 4 0x50 1 34  2 3ff                                                                                              
i2cwr: W to addr 0x50, reg 0x34, val=0x000003ff (2 bytes)                                                               
% i2crr 4 0x50 1 34 2                                                                                                   
i2cr: add: 0x50, reg 0x34: value 0x00000000 (2 bytes)
```
it should resemble the purpose of this code if it works, and I was not sure why we set 0x3ff as a value of this register when it's off when this value seems to flip 10 bits up rather than 8 bits of this register.

pwittich, 2022-10-14:
nice catch. It looks like the `cdr` and `rcvr` commands have the same issue, as you say. 
The way to test this command is to look at the temperature; once you turn off the transmitters the ff temperatures always go down a few degrees.

pwittich, 2022-10-14:
I'm actually really surprised none of our code checkers noticed this, it's unreachable code, isn't it?

pwittich, 2022-10-14:
If I add `-Wunreachable-code` to both clang and gcc builds clang does find it (gcc does not.) 
``` 
CC    commands/SensorControl.c
/Users/wittich/src/apollo_cm_mcu/projects/cm_mcu/commands/SensorControl.c:887:13: error: code will never be executed [-Werror,-Wunreachable-code]
        if (strncmp(argv[2], "on", 2) == 0) {
            ^~~~~~~
/Users/wittich/src/apollo_cm_mcu/projects/cm_mcu/commands/SensorControl.c:879:13: error: code will never be executed [-Werror,-Wunreachable-code]
        if (strncmp(argv[2], "on", 2) == 0) {
            ^~~~~~~
/Users/wittich/src/apollo_cm_mcu/projects/cm_mcu/commands/SensorControl.c:871:13: error: code will never be executed [-Werror,-Wunreachable-code]
        if (strncmp(argv[2], "on", 2) == 0) {
            ^~~~~~~
3 errors generated.
```

pwittich, 2022-10-18:
[image]
Turns out that codacy noticed this for one of the unreachable code entries, but somehow we missed it.

closed 2022-10-19 by pwittich