apollo-lhc/cm_mcu #156: FF part number seems to read into un-initialized territory

pwittich, 2022-10-27:
Are these strings null-terminated? 
```
202210271726 SRV INF LocalTasks.c:648:Getting Firefly 4-ch part (FPGA1): CRRNBY12024123M ,T 
:202210271726 SRV INF LocalTasks.c:682:Getting Firefly 4-ch part (FPGA2) : CRRNBY12024123M  
```

pkotamnives, 2022-10-28:
Hmm. which board did you test the FW on? I would like to reproduce the issue and investigate more.

pwittich, 2022-10-28:
Seems to be this on all of the ones?

pkotamnives, 2022-10-28:
I will investigate this on Apollo205 over the weekend with the master branch but I have never seen this string on Apollo214, 213 and CM202.

pwittich, 2022-10-28:
It’s a CERN-B part. I think the printout is from apollo205. My concern was that it appears not to be null terminated.

On Oct 28, 2022, at 17:34, Peace Kotamnives ***@***.***> wrote:

﻿

I will investigate this on Apollo205 over the weekend with the master branch but I have never seen this string on Apollo214, 213 and CM202.

pkotamnives, 2022-10-30:
you're right. It was not null-terminated before. I fixed that in the `fix_misc_bugs` branch.

closed 2022-11-04 by pkotamnives