apollo-lhc/cm_mcu #158: Snapshot clearall function

rzouCERN, 2022-10-31:
**Is your feature request related to a problem? Please describe.**
There are a lot of LGA80Ds on our board. The snapshot on each page of the device comes with a factory value that needs to be cleared first in order for it to work. It gets quite tedious to clear them up one by one. 

**Describe the solution you'd like**
A `snapshot all 1` function would be very helpful. It could also be helpful to be able to read them all at once with `snapshot all 0`.

pkotamnives, 2022-11-09:
hi @rzouCERN , what are some issues remaining with these commands ? Have I reproduced the issue from the snippet below? 

```
% snapshot all 1                                                                                                        
snapshot: page 0 of device 3V3/1V8                                                                                      
VIN  = 11.94                                                                                                            
VOUT = 1.80                                                                                                             
IOUT = 10.17                                                                                                            
IOUT MAX = 10.61                                                                                                        
duty cycle = 14.97                                                                                                      
TEMP = 37.69                                                                                                            
switching freq = 800.00                                                                                                 
VOUT  STATUS: 0x00                                                                                                      
IOUT  STATUS: 0x00                                                                                                      
INPUT STATUS: 0x00                                                                                                      
TEMP  STATUS: 0x00                                                                                                      
CML   STATUS: 0x00                                                                                                      
MFR   STATUS: 0x00                                                                                                      
flash STATUS: 0xff                                                                                                      
% snapshot all 0                                                                                                        
snapshot: page 0 of device 3V3/1V8                                                                                      
VIN  = 11.92                                                                                                            
VOUT = 1.81                                                                                                             
IOUT = 10.31                                                                                                            
IOUT MAX = 10.61                                                                                                        
duty cycle = 14.98                                                                                                      
TEMP = 37.13                                                                                                            
switching freq = 800.00                                                                                                 
VOUT  STATUS: 0x00                                                                                                      
IOUT  STATUS: 0x00                                                                                                      
INPUT STATUS: 0x00                                                                                                      
TEMP  STATUS: 0x00                                                                                                      
CML   STATUS: 0x00                                                                                                      
MFR   STATUS: 0x00                                                                                                      
flash STATUS: 0xff                                                                                                      
% snapshot 10 0                                                                                                         
snapshot: page 0 of device F1VCCINT1                                                                                    
VIN  = 6.22                                                                                                             
VOUT = 0.00                                                                                                             
IOUT = 0.00                                                                                                             
IOUT MAX = 0.00                                                                                                         
duty cycle = 0.00                                                                                                       
TEMP = 23.03                                                                                                            
switching freq = 457.00                                                                                                 
VOUT  STATUS: 0x00                                                                                                      
IOUT  STATUS: 0x00                                                                                                      
INPUT STATUS: 0x30                                                                                                      
[... cut ...]
INPUT STATUS: 0x30                                                                                                      
TEMP  STATUS: 0x00                                                                                                      
CML   STATUS: 0x00                                                                                                      
MFR   STATUS: 0x00                                                                                                      
flash STATUS: 0x00                                                                                                      
%         
```

pwittich, 2022-11-09:
hi @rzouCERN if we implement an `all` function it would probably overwhelm the output in minicom, if we then dump the registers at the same time. Were you thinking to just do a reset-only function that you could apply 'all' to?

closed 2023-03-30 by pkotamnives