apollo-lhc/cm_mcu #166: Add additional alarm tasks

pwittich, 2022-11-09:
Right now there is a `GenericAlarmTask` with callbacks. The only implementation of that task is for temperature out of range.

We should probably consider an additional alarm task for voltages out of range, and current out of range.

pkotamnives, 2022-11-16:
hi @pwittich , 

I have a few questions before implementing this task. Would you mind to answer them? 

- Should I create two new variables of type GenericAlarmParams_t for voltages and currents or create new set of files dedicating for each and add those two task files as two new tasks to cm_mcu.c ? 
- With the two main devices to monitor, I can tell from looking at the code that certain registers and variables are missing. Here is the list I have, and could you point me to where I can get them? 

1. **FFs** : missing registers for monitoring volatges and currents. Can I get them from the FF datasheets you gave for both 12-ch and 4-ch ones? 
2. **DCDC** : there are a long commands/registers already in [this](https://github.com/apollo-lhc/cm_mcu/blob/45cd2c0749274a1e839ab1c8602f1da0db155e30/projects/cm_mcu/LocalTasks.c#L257-L278). Which are the VOUT and IOUT registers to get their outputs from. 
3. **TM4C** : How to get new ADC_INFO_*_ENTRY for voltage and current ? and Can I still use the function `getADCvalue(ADC_INFO_*_ENTRY)` for this? 
4.  **FPGAs** : same issue as FFs, missing registers for monitoring volatges and currents but I don't recall that I have datasheets for FPGAs. Can you provide ones here?

- What would be the voltage and current thresholds for these devices? 

Thanks!

pwittich, 2022-11-17:
We can only monitor currents and voltages in the power supply modules or the currents directly measured by the TM4C ADC. 
from the LGA80D supplies we want `READ_IOUT` and `READ_VOUT`.
yes `getADCvalue` is a good place to start.

pkotamnives, 2022-11-18:
> 

For the currents measured by the TM4C ADC, which element in [this list](https://github.com/apollo-lhc/cm_mcu/blob/e37ed20786dcadc9400ed00fe749f7033fe15979/projects/cm_mcu/ADCMonitorTask.c#L106-L127) does it correspond to? I know that we currently read the temperature from the last element.

CRStrohman, 2022-11-18:
Currents are digitized in channels 13 through 17. I think the names of 
the 5 power sources being measured should be self-explanatory.

Charlie

closed 2023-03-09 by pkotamnives