apollo-lhc/cm_mcu #174: CM Inventory Information on Additional Registers

alpakpinar, 2023-01-25:
**Is your feature request related to a problem? Please describe.**
This is not related to a specific problem, but it would be really handy for inventory monitoring purposes to have several CM inventory values being written to `PL_MEM` registers. The eventual goal is to add the corresponding information to our database as well, and use our inventory monitoring software to read those registers under `PL_MEM`, and compare with the information on the database.

**Describe the solution you'd like**
It would be great if the following values could be written to the available `PL_MEM` registers:

* The CM revision, type (e.g. Cornell vs MPI) and serial number
* Data representing which Fireflies are installed on this CM. 
    * This could be something like a list of Firefly serial numbers and their types, in whichever way it is easiest to implement on the MCU FW side.
* Data representing which FPGAs are installed on this CM. 
    * This could be something like a list of FPGA serial numbers, in whichever way it is easiest to implement on the MCU FW side.

**Describe alternatives you've considered**
This is the only way we've so far discussed, as it is the easiest for our inventory software to read registers via `BUTool` CLI. But please let us know (especially since I'm not very familiar with fireflies & CM data in general), if you have alternative options in mind. 

**Additional context**
Thanks for considering this request! Apologies, I thought I opened this with issue #173 earlier (as the second issue discussed in Apollo meeting on Jan24). I'll include @dgastler here as the original requester of the feature.

pwittich, 2023-01-25:
@alpakpinar how is this different from #173 ?

alpakpinar, 2023-01-26:
@pwittich I believe it is different from #173 in the sense that here we would want the CM to write data to `PL_MEM.CM` registers on the Zynq so that we can read them using the `BUTool` interface. #173 was requested instead for us to create additional sensors on the IPMC, which can be read out from the `ipmitool sensor` command, so that those sensors can be accessed whether the Zynq is booted up or not.

For the IPMC sensors (#173), we wanted to add three new sensors representing:

* CM type
* CM revision
* CM serial number

In this issue, we'd like to have registers for the information above, but also add additional registers storing information for each FPGA and firefly installed on the CM. The presence of these registers will help us greatly for the online software and monitoring applications (e.g. keeping track of which fireflies are installed via running Apollo-HERD).

I'm currently discussing with Dan about the new registers to add to the address table for this purpose, and I can give you a draft XML of what we're thinking as soon as possible.

pwittich, 2023-01-27:
When you write ‘PL_MEM.CM’ I presume you are referring to the push data stream I send to the Zynq via a UART.  I don’t write anything into registers on the Zynq; I just push a data stream to the zynq.

pwittich, 2023-01-27:
Seems a bit excessive to send static data like that at a few tens of Hz tbh. We should maybe instead think of different types of data streams.

alpakpinar, 2023-01-27:
Ah yes, that's what I meant indeed (thanks for clarifying, I didn't know much about how this worked on the CM side). So do you only need a list of data to push and the memory addresses on Zynq to push them?

I see, indeed, would it be possible/easy to open a second stream (maybe another UART?) with much lower frequency?

pwittich, 2023-05-24:
@alpakpinar @dgastler @pkotamnives 
I propose to send a bit mask for which ff are mounted and a second bit mask for if they are CERN-B or 25G for the 12 channel parts. Agreed?

closed 2023-11-14 by pwittich