apollo-lhc/cm_mcu #215: BUG - Programming MCU FW + Resetting UIO Daemon

Gianco99, 2024-03-19:
Steps to reproduce the error include:

Install MCU FW and set-up address tables in /fw/CM as described here - https://apollo-lhc.gitlab.io/SM_Firmware/CM_MCU/Installation/

Reboot UIO Daemon to find new MCU address tables as described in the second note here - https://apollo-lhc.gitlab.io/SM_Software/linux-daemons/#daemons

This sequence of commands appears to be successful; however, it brings the Zynq CPU down. A simple reset of the Zynq resolves the issue, and MCU FW registers can be read appropriately.

Gianco99, 2024-03-19:
Linked to - https://gitlab.com/apollo-lhc/soc-tools/bootup-daemon/-/issues/8

pwittich, 2024-03-19:
which version of the MCU FW did you install? 
FWIW, I don't think this is a MCU bug, this sounds like a zynq bug?

Gianco99, 2024-03-22:
This was using - v0.99.6

pwittich, 2024-06-11:
@Gianco99 can this be closed? I don't think this MCU related.

closed 2024-06-13 by pwittich