apollo-lhc/cm_mcu #216: Per-Channel Monitoring Information for Fireflies

Gianco99, 2024-03-22:
For Shep/Herd, the ability to access the per-channel information of the FF is important to be able to debug any faulty links during a link/slice test. 

Currently we have the IsPresent bits that let us know which FFs are on the board. However, a similar type of bit letting us know which channels are enabled/disabled on a FF would also be very useful.

I understand it is a bit convoluted to implement a way to change which channels on a FF are enabled/disabled, so this isn't as pressing. Just being able to read a register - similar to IsPresent - that lets us know which channels are enabled/disabled would be very useful for the link/slice test debugging.

pwittich, 2024-05-13:
@Gianco99 we need to solve #210 to make this easy to use.

pwittich, 2024-06-12:
@Gianco99 #221 adds this as a CLI command; I can add it now to push to the zynq.

pwittich, 2024-07-17:
```
        CM_FF_MON| OPTPOW| TEMP_C| TXDISABLED|
               --|-------|-------|-----------|
       F1_P1_RX12|   3436|     31|          0|
       F1_P1_TX12|      0|      0|          0|
       F1_P2_RX12|   3288|     33|          0|
       F1_P2_TX12|      0|     29|          0|
       F1_P3_RX12|   2443|     36|          0|
       F1_P3_TX12|      0|     23|          0|
      F1_P4_XCVR4|      0|      0|          0|
      F1_P5_XCVR4|      0|      0|          0|
      F1_P6_XCVR4|  62025|     35|          0|
      F1_P7_XCVR4|      0|      0|          0|
       F2_P1_RX12|   1271|     36|          0|
       F2_P1_TX12|      0|     33|          0|
       F2_P2_RX12|   3090|     34|          0|
       F2_P2_TX12|      0|     30|          0|
       F2_P3_RX12|   2895|     30|          0|
       F2_P3_TX12|      0|     30|          0|
      F2_P4_XCVR4|      0|      0|          0|
      F2_P5_XCVR4|      0|      0|          0|
      F2_P6_XCVR4|  62664|     33|          0|
      F2_P7_XCVR4|      0|      0|          0|
```
(run `ff xmit off all` in other window)
```
>status 9 CM_FF_MON
Process done
        CM_FF_MON| OPTPOW| TEMP_C| TXDISABLED|
               --|-------|-------|-----------|
       F1_P1_RX12|  60748|     36|          0|
       F1_P1_TX12|      0|      0|       4095|
       F1_P2_RX12|   3288|     35|          0|
       F1_P2_TX12|      0|     30|       4095|
       F1_P3_RX12|   2443|     38|          0|
       F1_P3_TX12|      0|     24|       4095|
      F1_P4_XCVR4|      0|      0|          0|
      F1_P5_XCVR4|      0|      0|          0|
      F1_P6_XCVR4|     75|     35|         15|
      F1_P7_XCVR4|      0|      0|          0|
       F2_P1_RX12|   2062|     41|          0|
       F2_P1_TX12|      0|     35|       4095|
       F2_P2_RX12|   8727|     38|          0|
       F2_P2_TX12|      0|     32|       4095|
       F2_P3_RX12|  56380|     33|          0|
       F2_P3_TX12|      0|     31|       4095|
      F2_P4_XCVR4|      0|      0|          0|
      F2_P5_XCVR4|      0|      0|          0|
      F2_P6_XCVR4|    300|     33|         15|
      F2_P7_XCVR4|      0|      0|          0|

```
Installed on 213 and 214 at the tif.

pwittich, 2024-07-17:
@Gianco99 I'm going to close this issue; please reopen if this is not what you had in mind.

closed 2024-07-17 by pwittich