apollo-lhc/cm_mcu #227: FF IS_PRESENT Registers Giving Strange Results

Gianco99, 2024-06-25:
The registers "PL_MEM_CM.F1_FFL12CH.IS_PRESENT" and "PL_MEM_CM.F1_FFLDAQ.IS_PRESENT" are supposed to be bitmasks, but are giving unexpected reads shown in the images below. Note the MCU version in the first image, consistent with the version returned by the MCU CL.

The address table was cross-checked to be consistent with the MCU version and still gave these unknown reads,

Going into the MCU CL through minicom shows two FFs connected (as expected), but their presence is not reflected in the IS_PRESENT registers.

[image]

[image]

pwittich, 2024-06-25:
@Gianco99 have you looked at the other bits 
https://github.com/apollo-lhc/cm_mcu/blob/049babb8157772bf833b578fd17938c4adf8e448/sm_cm_config/data/PL_MEM_CM_rev2.yml#L244-L245
Do they look sensible?

pwittich, 2024-06-25:
Also, it looks like the `IS_FF25Gbs` is 6 bits for the 12 channel parts, i.e., one each for Tx and Rx times 3 different sets of devices. Do you care if I change that to be 3 bits only (i.e., don't allow mismatched 25 Gb and CERN-B)?

pwittich, 2024-06-26:
on apollo204
```
>read PL_MEM_CM.F*_FF*.IS_*
                   PL_MEM_CM.F1_FFL12CH.IS_FF25Gbs: 0x00000000
                   PL_MEM_CM.F1_FFL12CH.IS_PRESENT: 0x00000000
                    PL_MEM_CM.F1_FFLDAQ.IS_FF25Gbs: 0x00000033
                    PL_MEM_CM.F1_FFLDAQ.IS_PRESENT: 0x0000000F
                   PL_MEM_CM.F2_FFL12CH.IS_FF25Gbs: 0x00000003
                   PL_MEM_CM.F2_FFL12CH.IS_PRESENT: 0x0000000F
                    PL_MEM_CM.F2_FFLDAQ.IS_FF25Gbs: 0x00000033
                    PL_MEM_CM.F2_FFLDAQ.IS_PRESENT: 0x0000000F
% ff_status
FF STATUS:
PRESENT:
F1_12: 0x33
F1_4 : 0x03
F2_12: 0x33
F2_4 : 0x0c
      F1_1  12 Tx: 0x28 3v8?(0)               F1_1  12 Rx: 0x00
      F1_2  12 Tx:   -- 3v8?(0)               F1_2  12 Rx:   --
      F1_3  12 Tx: 0x02 3v8?(1)               F1_3  12 Rx: 0x02
      F1_4 4 XCVR: 0x00
      F1_5 4 XCVR: 0x00
      F1_6 4 XCVR:   --
      F1_7 4 XCVR:   --
      F2_1  12 Tx: 0x02 3v8?(1)               F2_1  12 Rx: 0x02
      F2_2  12 Tx:   -- 3v8?(0)               F2_2  12 Rx:   --
      F2_3  12 Tx: 0x28 3v8?(0)               F2_3  12 Rx: 0x00
      F2_4 4 XCVR:   --
      F2_5 4 XCVR:   --
      F2_6 4 XCVR: 0x00
      F2_7 4 XCVR: 0x00
% ff_dump_names
ff_dump_names: ID registers
      F1_1  12 Tx: Y1214020011221             F1_1  12 Rx: YR214020011221  
      F1_2  12 Tx: --------------             F1_2  12 Rx: --------------
      F1_3  12 Tx: ECUOT12251000513           F1_3  12 Rx: ECUOR12251000513
      F1_4 4 XCVR: B0425040011201  
      F1_5 4 XCVR: B0425040011201  
      F1_6 4 XCVR: --------------
      F1_7 4 XCVR: --------------
      F2_1  12 Tx: ECUOT12251000513           F2_1  12 Rx: ECUOR12251000513
      F2_2  12 Tx: --------------             F2_2  12 Rx: --------------
      F2_3  12 Tx: Y1214020011221             F2_3  12 Rx: YR214020011221  
      F2_4 4 XCVR: --------------
      F2_5 4 XCVR: --------------
      F2_6 4 XCVR: B0425040011201  
      F2_7 4 XCVR: B0425040011201  
```
so it looks like the 25 Gbps and the present are mixed up. The present is shown in `ff_status` on the MCU

Gianco99, 2024-06-26:
Ah okay, I have been using release [v0.99.6](https://github.com/apollo-lhc/cm_mcu/releases/tag/v0.99.6) on Apollo212 and TIF213 before.

Is this an issue with a newer release or does something else trigger it?

pwittich, 2024-06-26:
I think even the  head is wrong, so I need to fix that. that was just my keeping track of what I've discovered.

closed 2024-06-28 by pwittich