apollo-lhc/cm_mcu #250: Implement per-channel FF monitoring when the MCU FW has this capability

furryboffin, 2025-02-06:


Gianco99, 2025-02-06:
We are already reading the `ff_optpow` information from registers accessible via BUTool. For example, reading the register `PL_MEM_CM.F1_P1_Rx12.Temp_C` every x amount of time to monitor this value.

However, we would also like access, for example, the information present when running `ff_optpow_dev 6` to also monitor. It would be best to be able to access this information in a similar way using BUTool to read the register from the Zynq.

```
% ff_optpow
FF average Optical Power (uW)
      F1_1  12 Tx:     ---                    F1_1  12 Rx:    12.72
      F1_2  12 Tx:     ---                    F1_2  12 Rx:    24.43
      F1_3  12 Tx:     ---                    F1_3  12 Rx:    30.90
      F1_4 4 XCVR:  1413.00
      F1_5 4 XCVR:  1507.75
      F1_6 4 XCVR:  1419.00
      F1_7 4 XCVR:  1324.25
      F2_1  12 Tx:     ---                    F2_1  12 Rx:    28.96
      F2_2  12 Tx:     ---                    F2_2  12 Rx:    37.79
      F2_3  12 Tx:     ---                    F2_3  12 Rx:    24.04
      F2_4 4 XCVR:  14      F2_5 4 XCVR:  1306.50
      F2_6 4 XCVR:  1442.50
      F2_7 4 XCVR:  1292.00
% ff_optpow_dev 6
FF F1_4 4 XCVR Optical Power (uW)
Ch 00:  1278.00
Ch 01:  1469.00
Ch 02:  1478.00
Ch 03:  1424.00
% 
```

pwittich, 2025-02-07:
@Gianco99 is this (optical power per channel) the only quantity you are looking for?

Gianco99, 2025-02-12:
These are the list of quantities we are interested in

- Received optical power - one per rx channel
- LOL (Loss Of Lock) - one per channel (each direction)
- LOS (Loss Of Signal) - one per channel (each direction on 4Ch parts, and 12Ch Rx part)
- Laser fault - one per tx channel

pwittich, 2025-02-12:
> These are the list of quantities we are interested in
> 
> * Received optical power - one per rx channel (25G and 10G 12Ch, 25G 4Ch)
This quantity does not exist on the 10G, 12 Ch parts.

Gianco99, 2025-02-13:
Memory Addresses from the SAMTEC Datasheets for 25G 4Ch parts are given below. Still looking for the 12Ch datasheet

25G 4Ch:

Measured Optical Power
Address Channel
- 34-35 Rx1 
- 36-37 Rx2 
- 38-39 Rx3
- 40-41 Rx4

Loss of Signal Alarms, address 3
Byte Channel
- 0 Rx1
- 1 Rx2
- 2 Rx3
- 3 Rx4
- 4 Tx1
- 5 Tx2
- 6 Tx3
- 7 Tx4

CDR Loss of Lock Alarms, address 5
Byte Channel
- 7 Tx4
- 6 Tx3
- 5 Tx2 
- 4 Tx1
- 3 Rx4
- 2 Rx3
- 1 Rx2
- 0 Rx1

Transmit Fault Alarms, address 4
Byte Channel

- 3 Tx4
- 2 Tx3
- 1 Tx2
- 0 Tx1

Gianco99, 2025-02-14:
Here are all the memory locations formatted as [Byte, Bit, Channel] I was able to compile from the different datasheets.

As you said, 12Ch CernBs do not have optical power, but I found a few more things missing from the 10G parts, namely:

Both LOS and LOL missing from the Tx 12Ch CernBs
LOL missing from the Rx 12Ch CernBs

For now, if you can just add the list below that would be great and I will look into why these are missing

###############################
###############################

25G 4Ch:

Measured Optical Power
Byte Channel
- 34-35 Rx1 
- 36-37 Rx2 
- 38-39 Rx3
- 40-41 Rx4

Loss of Signal Alarms
Byte Bit Channel
- 3 0 Rx1
- 3 1 Rx2
- 3 2 Rx3
- 3 3 Rx4
- 3 4 Tx1
- 3 5 Tx2
- 3 6 Tx3
- 3 7 Tx4

CDR Loss of Lock Alarms
Byte Bit Channel
- 5 7 Tx4
- 5 6 Tx3
- 5 5 Tx2 
- 5 4 Tx1
- 5 3 Rx4
- 5 2 Rx3
- 5 1 Rx2
- 5 0 Rx1

Transmit Fault Alarms
Byte Bit Channel

- 4 3 Tx4
- 4 2 Tx3
- 4 1 Tx2
- 4 0 Tx1

###############################
###############################

25G 12Ch Tx

Loss of Signal Alarms
Byte Bit Channel
- 7 3 Tx11
- 7 2 Tx10
- 7 1 Tx9
- 7 0 Tx8
- 8 7 Tx7
- 8 6 Tx6
- 8 5 Tx5
- 8 4 Tx4
- 8 3 Tx3
- 8 2 Tx2
- 8 1 Tx1
- 8 0 Tx0

Loss of Lock
Byte Bit Channel
- 20 3 Tx11
- 20 2 Tx10
- 20 1 Tx9
- 20 0 Tx8
- 21 7 Tx7
- 21 6 Tx6
- 21 5 Tx5
- 21 4 Tx4
- 21 3 Tx3
- 21 2 Tx2
- 21 1 Tx1
- 21 0 Tx0

Transmit Fault Alarms
Byte Bit Channel
- 9 3 Tx11
- 9 2 Tx10
- 9 1 Tx9
- 9 0 Tx8
- 10 7 Tx7
- 10 6 Tx6
- 10 5 Tx5
- 10 4 Tx4
- 10 3 Tx3
- 10 2 Tx2
- 10 1 Tx1
- 10 0 Tx0

##############################
##############################

25G 12Ch Rx

Loss of Signal Alarms
Byte Bit Channel
- 7 3 Rx11
- 7 2 Rx10
- 7 1 Rx9
- 7 0 Rx8
- 8 7 Rx7
- 8 6 Rx6
- 8 5 Rx5
- 8 4 Rx4
- 8 3 Rx3
- 8 2 Rx2
- 8 1 Rx1
- 8 0 Rx0

Loss of Lock
Byte Bit Channel
- 20 3 Rx11
- 20 2 Rx10
- 20 1 Rx9
- 20 0 Rx8
- 21 7 Rx7
- 21 6 Rx6
- 21 5 Rx5
- 21 4 Rx4
- 21 3 Rx3
- 21 2 Rx2
- 21 1 Rx1
- 21 0 Rx0

Measured Optical Power
Byte Channel
- 206-207 Rx11
- 208-209 Rx10
- 210-211 Rx9
- 212-213 Rx8
- 214-215 Rx7
- 216-217 Rx6
- 218-219 Rx5
- 220-221 Rx4
- 222-223 Rx3
- 224-225 Rx2
- 226-227 Rx1
- 228-229 Rx0

##############################
##############################

10G 12Ch Tx

Transmit Fault Alarms
Byte Bit Channel
- 9 3 Tx11
- 9 2 Tx10
- 9 1 Tx9
- 9 0 Tx8
- 10 7 Tx7
- 10 6 Tx6
- 10 5 Tx5
- 10 4 Tx4
- 10 3 Tx3
- 10 2 Tx2
- 10 1 Tx1
- 10 0 Tx0

##############################
##############################

10G 12Ch Rx

Loss of Signal Alarms
Byte Bit Channel
- 7 3 Rx11
- 7 2 Rx10
- 7 1 Rx9
- 7 0 Rx8
- 8 7 Rx7
- 8 6 Rx6
- 8 5 Rx5
- 8 4 Rx4
- 8 3 Rx3
- 8 2 Rx2
- 8 1 Rx1
- 8 0 Rx0

pwittich, 2025-04-25:
```>status 3 CM_OPTPOW12_MON
Process done
  CM_OPTPOW12_MON| OPTPOW_CH0| OPTPOW_CH1| OPTPOW_CH10| OPTPOW_CH11| OPTPOW_CH2| OPTPOW_CH3| OPTPOW_CH4| OPTPOW_CH5| OPTPOW_CH6| OPTPOW_CH7| OPTPOW_CH8| OPTPOW_CH9|
               --|-----------|-----------|------------|------------|-----------|-----------|-----------|-----------|-----------|-----------|-----------|-----------|
      FF1_P1_RX12|          0|          0|          40|           0|          0|          0|          0|          0|         45|         12|          0|         53|
      FF1_P2_RX12|         71|          6|           0|          80|          0|          0|         33|          0|         50|          0|         21|         28|
      FF1_P3_RX12|          0|          0|           0|           0|          0|          0|          0|          0|          0|          0|          0|          0|
      FF1_P4_RX12|         46|         14|          37|          36|         31|          0|         43|         78|          0|          0|          0|         82|
      FF2_P1_RX12|         29|         45|          39|          42|         61|          0|         45|          0|          0|         35|          0|         47|
      FF2_P2_RX12|         59|         29|           0|           0|         14|          5|         67|          9|         63|         85|         23|         94|
      FF2_P3_RX12|          0|          0|           0|           0|          0|          0|          0|          0|          0|          0|          0|          0|
      FF2_P4_RX12|         17|         33|           0|          31|         13|          0|         45|         34|         73|          5|         23|         10|

SW VER: -1
>status 3 CM_OPTPOW4_MON
Process done
   CM_OPTPOW4_MON| OPTPOW_CH0| OPTPOW_CH1| OPTPOW_CH2| OPTPOW_CH3|
               --|-----------|-----------|-----------|-----------|
     FF1_P5_XCVR4|       1439|       1510|       1491|       1419|
     FF1_P6_XCVR4|       1413|       1537|       1377|       1396|
     FF2_P5_XCVR4|       1642|       1425|       1524|       1491|
     FF2_P6_XCVR4|       1291|       1329|       1287|       1423|

SW VER: -1
```

pwittich, 2025-04-26:
Updated column headers, and this is for Rev2 (above was for rev3). These are CERN-B parts so nothing on the 12-wide.
```
  CM_OPTPOW12_MON| CH0| CH1| CH10| CH11| CH2| CH3| CH4| CH5| CH6| CH7| CH8| CH9|
               --|----|----|-----|-----|----|----|----|----|----|----|----|----|
      FF1_P1_RX12|   0|   0|    0|    0|   0|   0|   0|   0|   0|   0|   0|   0|
      FF1_P2_RX12|   0|   0|    0|    0|   0|   0|   0|   0|   0|   0|   0|   0|
      FF1_P3_RX12|   0|   0|    0|    0|   0|   0|   0|   0|   0|   0|   0|   0|
      FF2_P1_RX12|   0|   0|    0|    0|   0|   0|   0|   0|   0|   0|   0|   0|
      FF2_P2_RX12|   0|   0|    0|    0|   0|   0|   0|   0|   0|   0|   0|   0|
      FF2_P3_RX12|   0|   0|    0|    0|   0|   0|   0|   0|   0|   0|   0|   0|

   CM_OPTPOW4_MON|  CH0|  CH1|  CH2|  CH3|
               --|-----|-----|-----|-----|
     FF1_P4_XCVR4| 1081|  995|  879| 1444|
     FF1_P5_XCVR4|    0|    3|    1|    1|
     FF1_P6_XCVR4|    0|    1|    0|    0|
     FF1_P7_XCVR4|    0|    4|    1|    0|
     FF2_P4_XCVR4| 1379| 1360| 1377| 1310|
     FF2_P5_XCVR4| 1416| 1244| 1273| 1266|
     FF2_P6_XCVR4| 1322| 1222| 1314| 1356|
     FF2_P7_XCVR4| 1179| 1251| 1203| 1246|
```

pwittich, 2025-04-26:
Will need to add something like this to BUMonitor.BU:
```
sensor=ApolloMonitor CM_OPTPOW4_MON graphite apolloSM $SN CM_OPTPOW4_MON 3
sensor=ApolloMonitor CM_OPTPOW12_MON graphite apolloSM $SN CM_OPTPOW12_MON 3
```

pwittich, 2025-04-28:
[image]
4 channel optical power

closed 2025-04-29 by pwittich